JanM321/esphome-lg-controller #101: 30 pin vs 38 pin ESP32

robkap, 2025-03-02:
Will my 30 pin ESP32 board work with the hardware-tiny PCB?  I have a https://www.amazon.com/dp/B08D5ZD528 .  Amazon's description is: 
"ESP-WROOM-32 ESP32 ESP-32S Development Board 2.4GHz Dual-Mode WiFi + Bluetooth Dual Cores Microcontroller Processor Integrated with Antenna RF AMP Filter AP STA Compatible with Arduino IDE"

d4m14ndx, 2025-03-02:
Yes, but you will need to use jumper wires to re-patch the pins correctly instead of inserting your devboard directly into the Tiny - soemthing like the male-female found here - https://www.amazon.com.au/120pcs-Multicoloured-Dupont-Breadboard-arduino/dp/B01EV70C78/

robkap, 2025-03-02:
Can you recommend a better ESP32 (from Amazon?) that won't require jumpers?