JanM321/esphome-lg-controller #123: The Feather S3 is not communicating with the PREMTB100 :(

Sakab001, 2025-08-30:
Hi everyone! Status update and a request for help.

First of all, I don’t have an air conditioner, but an LG heat recovery ventilator (LG LZ-H035GBA5) and a PREMTB100.
The Feather S3 and its corresponding board have arrived. I connected it, flashed it, and added it to Home Assistant. It connects to Wi-Fi, Home Assistant recognizes it, and no errors are shown. I can toggle the buttons, but nothing happens.

I tried all the options: heating, cooling, fan only (low, medium, high fan speed) – but nothing changes on the original controller.

What I noticed is that when I first connected it while it was powered on, the display changed and looked like this:
[(image)](https://i.ibb.co/tpF4sVYD/viber-k-p-2025-08-30-15-12-58-424.jpg)

Then, after I power-cycled it with the connection in place, the original display returned, but its color seemed to change:
[(image)](https://i.ibb.co/7dNWmBG8/viber-k-p-2025-08-30-15-12-58-444.jpg)

Another thing I noticed with the Feather S3 is that once it was powered only by the PREMTB100, not all 4 LEDs were blinking—only the two closer to the USB-C port. The two on the back did not light up or blink.

Also, I didn’t pay attention before, so I’m not sure if it was like this earlier, but now this chip gets extremely hot—so hot I can’t keep my finger on it. I’m not sure if that’s normal. Naturally, when it’s connected to the PREMTB100, it does not get any additional power from a separate Type-C adapter.
[(image)](https://i.ibb.co/tTJySM8w/feather.jpg)

Other than that, it doesn’t show any errors, it connects to Wi-Fi, and Home Assistant recognizes it.

Here are my YAML files:
[(image)](https://i.ibb.co/20FqL4X6/yaml.jpg)
[(image)](https://i.ibb.co/8nHCzKqm/baseyaml.jpg)

What did I do wrong? :/

Sakab001, 2025-08-30:
I don't know if this helps, but I was able to extract this log:
ESP-ROM:esp32s3-20210327

[I][logger:165]: Log initialized

[C][safe_mode:082]: There have been 0 suspected unsuccessful boot attempts

[D][esp32.preferences:142]: Writing 1 items: 0 cached, 1 written, 0 failed

[I][app:090]: Running through setup()

[C][component:164]: Setup uart took 0ms

[D][climate:396]: 'LG Living Room' - Sending state:

[D][climate:399]:   Mode: FAN_ONLY

[D][climate:425]:   Target Temperature: 30.00°C

[D][switch:0[D][lg-controller:1358]: update

[E][lg-controller:750]: unknown fan mode, using Medium

[E][lg-controller:774]: unknown swing mode

[D][lg-controller:853]: sending A8.2B.00.00.00.00.0F.14.40.00.80.00.E3 (13)

[W][component:423]: lg_controller.climate took a long time for an operation (510                                                                                                                                                                                                                                              ms)

[W][component:424]: Components should block for at most 30 ms

[W][component:423]: api took a long time for an operation (512 ms)

[W][component:424]: Components should block for at most 30 ms

[D][lg-controller:1358]: update

[E][lg-controller:876]: Unexpected missing previous CA/AA message

[D][lg-controller:1358]: update

JanM321, 2025-08-31:
Your unit may use a protocol that's different from the one used for AC units. There's some information in #33

From your log file it looks like the controller is sending messages but not receiving anything. I'd expect to see at least some data coming from the PREMTB100 if it's not set to slave mode. You should double check the data wire (the middle one) is connected properly.

Sakab001, 2025-08-31:
Hi, thank you for your reply. I will check the signal cable, but since the display has also changed, that shouldn’t be an issue. I looked at the post, and the author of the post has the same type of ventilator as I do, only theirs is the smaller one, while mine is the higher-performance version. I would only like to integrate the commands listed in this table into Home Assistant. Are these the commands that are included in the basic protocol file that has been uploaded? How can I read what code the PREMTB is sending? Because if I can read those 6–7 codes, I will extract them and integrate them into my own protocol file.

[image]

JanM321, 2025-09-01:
> Should I be seeing the command line sent by the PREMTB here?

The log you posted yesterday, that's where you should see messages that are sent/received. I only use the esphome command line and there you can use `esphome logs ..` to watch new log messages.

> Once I have the codes, exactly which file should I open and modify for the command line?

lg-controller.h is where the AC protocol is implemented.

Sakab001, 2025-10-20:
Hi everyone, after a short break, I'm getting back

into decoding the commands of the LG heat recovery ventilator with renewed energy. I installed it, and this is the type of log I'm getting. This is completely different from the commands described in the documentation. This is specifically the machine's status message. What could be the problem, what am I doing wrong, and how can I read the actual command sequence? (PREMTB100)

`[22:44:14.532][I][remote.pronto:231]: Received Pronto: data= [22:44:14.532][I][remote.pronto:233]: 0000 006D 0002 0000 0301 0180 0301 0300 [22:44:14.554][I][remote.pronto:231]: Received Pronto: data= [22:44:14.554][I][remote.pronto:233]: 0000 006D 0001 0000 0301 0300 [22:44:14.631][I][remote.pronto:231]: Received Pronto: data= [22:44:14.633][I][remote.pronto:233]: 0000 006D 0001 0000 0181 0300 [22:44:14.732][I][remote.pronto:231]: Received Pronto: data= [22:44:14.735][I][remote.pronto:233]: 0000 006D 0002 0000 0301 0180 0181 0300 [22:44:14.800][I][remote.pronto:231]: Received Pronto: data= [22:44:14.800][I][remote.pronto:233]: 0000 006D 0001 0000 0180 0300 [22:44:14.822][I][remote.pronto:231]: Received Pronto: data= [22:44:14.822][I][remote.pronto:233]: 0000 006D 0001 0000 0180 0300 [22:44:14.927][I][remote.pronto:231]: Received Pronto: data= [22:44:14.929][I][remote.pronto:233]: 0000 006D 0001 0000 0180 0300 [22:44:15.039][I][remote.pronto:231]: Received Pronto: data= [22:44:15.039][I][remote.pronto:233]: 0000 006D 0001 0000 0181 0300 [22:44:15.123][I][remote.pronto:231]: Received Pronto: data= [22:44:15.123][I][remote.pronto:233]: 0000 006D 0001 0000 0181 0300 [22:44:15.224][I][remote.pronto:231]: Received Pronto: data= [22:44:15.225][I][remote.pronto:233]: 0000 006D 0001 0000 0181 0300 [22:44:15.329][I][remote.pronto:231]: Received Pronto: data= [22:44:15.330][I][remote.pronto:233]: 0000 006D 0001 0000 0181 0300 [22:44:15.420][I][remote.pronto:231]: Received Pronto: data= [22:44:15.424][I][remote.pronto:233]: 0000 006D 0001 0000 0181 0300 [22:44:15.526][I][remote.pronto:231]: Received Pronto: data= [22:44:15.527][I][remote.pronto:233]: 0000 006D 0001 0000 0181 0300 [22:44:15.633][I][remote.pronto:231]: Received Pronto: data= [22:44:15.634][I][remote.pronto:233]: 0000 006D 0002 0000 0180 0181 0302 [22:44:15.721][I][remote.pronto:231]: Received Pronto: data= [22:44:15.722][I][remote.pronto:233]: 0000 006D 0001 0000 0181 0302 [22:44:16.030][I][remote.pronto:231]: Received Pronto: data= [22:44:16.031][I][remote.pronto:233]: 0000 006D 0002 0000 0301 0180 0301 0300 [22:44:16.067][I][remote.pronto:231]: Received Pronto: data= [22:44:16.068][I][remote.pronto:233]: 0000 006D 0001 0000 0301 0300 [22:44:16.121][I][remote.pronto:231]: Received Pronto: data= [22:44:16.122][I][remote.pronto:233]: 0000 006D 0001 0000 0181 0300 [22:44:16.229][I][remote.pronto:231]: Received Pronto: data= [22:44:16.230][I][remote.pronto:233]: 0000 006D 0002 0000 0302 0181 0181 0300 [22:44:16.301][I][remote.pronto:231]: Received Pronto: data= [22:44:16.302][I][remote.pronto:233]: 0000 006D 0001 0000 0180 0300 [22:44:16.322][I][remote.pronto:231]: Received Pronto: data= [22:44:16.324][I][remote.pronto:233]: 0000 006D 0001 0000 0180 0300 [22:44:16.428][I][remote.pronto:231]: Received Pronto: data= [22:44:16.428][I][remote.pronto:233]: 0000 006D 0001 0000 0181 0300 [22:44:16.535][I][remote.pronto:231]: Received Pronto: data= [22:44:16.535][I][remote.pronto:233]: 0000 006D 0001 0000 0180 0300 [22:44:16.627][I][remote.pronto:231]: Received Pronto: data= [22:44:16.627][I][remote.pronto:233]: 0000 006D 0001 0000 0181 0300 [22:44:16.731][I][remote.pronto:231]: Received Pronto: data= [22:44:16.733][I][remote.pronto:233]: 0000 006D 0001 0000 0181 0300 [22:44:16.821][I][remote.pronto:231]: Received Pronto: data= [22:44:16.822][I][remote.pronto:233]: 0000 006D 0001 0000 0181 0300 [22:44:16.927][I][remote.pronto:231]: Received Pronto: data= [22:44:16.928][I][remote.pronto:233]: 0000 006D 0001 0000 0180 0300 [22:44:17.034][I][remote.pronto:231]: Received Pronto: data= [22:44:17.034][I][remote.pronto:233]: 0000 006D 0001 0000 0181 0300 [22:44:17.139][I][remote.pronto:231]: Received Pronto: data= [22:44:17.140][I][remote.pronto:233]: 0000 006D 0002 0000 0181 0180 0302 [22:44:17.229][I][remote.pronto:231]: Received Pronto: data= [22:44:17.230][I][remote.pronto:233]: 0000 006D 0001 0000 0181 0302 [22:44:17.642][I][LG-BUS:063]: Várjuk a CN-REMO busz jeleit... [22:44:17.672][I][remote.pronto:231]: Received Pronto: data= [22:44:17.673][I][remote.pronto:233]: 0000 006D 0005 0000 0181 0181 0301 0180 0301 0181 0180 0181 0181 0300 [22:44:17.747][I][remote.pronto:231]: Received Pronto: data= [22:44:17.748][I][remote.pronto:233]: 0000 006D 0002 0000 0180 0181 0300 [22:44:17.829][I][remote.pronto:231]: Received Pronto: data= [22:44:17.829][I][remote.pronto:233]: 0000 006D 0001 0000 0181 0300 [22:44:17.924][I][remote.pronto:231]: Received Pronto: data= [22:44:17.925][I][remote.pronto:233]: 0000 006D 0001 0000 0180 0300 [22:44:18.030][I][remote.pronto:231]: Received Pronto: data= [22:44:18.031][I][remote.pronto:233]: 0000 006D 0001 0000 0180 0300 [22:44:18.124][I][remote.pronto:231]: Received Pronto: data= [22:44:18.125][I][remote.pronto:233]: 0000 006D 0001 0000 0181 0300 [22:44:18.258][I][remote.pronto:231]: Received Pronto: data= [22:44:18.264][I][remote.pronto:233]: 0000 006D 0002 0000 0180 0181 0301 0300 [22:44:18.297][I][remote.pronto:231]: Received Pronto: data= [22:44:18.298][I][remote.pronto:233]: 0000 006D 0001 0000 0181 0 [... cut ...] 0 006D 0001 0000 0180 0300 [22:44:20.058][I][remote.pronto:231]: Received Pronto: data= [22:44:20.059][I][remote.pronto:233]: 0000 006D 0002 0000 0180 0181 0300 [22:44:20.146][I][remote.pronto:231]: Received Pronto: data= [22:44:20.146][I][remote.pronto:233]: 0000 006D 0001 0000 0180 0302 [22:44:20.236][I][remote.pronto:231]: Received Pronto: data= [22:44:20.236][I][remote.pronto:233]: 0000 006D 0001 0000 0181 0300 [22:44:22.643][I][LG-BUS:063]: Várjuk a CN-REMO busz jeleit... [22:44:27.644][I]`