JanM321/esphome-lg-controller #124: Seed XIAO ESP32-C6 target

ireun, 2025-09-10:
Hi!

I've started my work of porting featherboard design to fit Seed Studio XIAO ESP32-C6:

Changes:
* TestPoints (GND, 3V3, 12V)
* Additional Cap C3 on 3V3 output of TPS82130
* Ability to mount XIAO directly on board 
* 2 layers only 
* Signal pin is in the middle of JST connector

[image]
[image]
[image]

Let me know what you think :)

ireun, 2025-09-22:
@martiinezz I have no idea, you may look at https://www.esphome.io/components/climate/climate_ir/ or record original IR remote using IR receiver, and relay these messages later :)

Vincentb19, 2025-11-02:
Hi @ireun, Have you ordered the PCB's? And if so, does it work? can you share the files you have made? :)

ireun, 2025-11-12:
@Vincentb19 Hi! I've published the files here: https://github.com/ireun/esphome-lg-controller/tree/main/hardware-SeeedXiaoC6

Keep in mind that I have not tested this, I have not prepared gerber files. Most importantly I will probably never gonna 'finish' this. I just 'think' it should work. :)